openLRSng/openLRSng-configurator #19: Cant connect my new HK RX to configurator

drak1nguard, 2015-01-13:
Hello to all....can someone help me.....
I bought from HB a new ORX 433 100mw and i am trying to flash it with new firmware!The usb drivers installed good as my pc tells me!Also i connect it as they told me in the support but when i hit the connect button on the configurator this message always pop out:
15:35:48 -- Serial port successfully opened with ID: 3
15:35:58 -- Start message not received within 10 seconds, disconnecting
15:35:58 -- Serial port successfully closed
15:36:06 -- Serial port successfully opened with ID: 4
15:36:17 -- Start message not received within 10 seconds, disconnecting
15:36:17 -- Serial port successfully closed

And dont let me flash it....can someone help me?

cTn-dev, 2015-01-14:
I would start by checking your VCC/GND/RX/TX/DTR connections, if they are all fine and wires aren't loose, try swapping the RX and TX wires, if that also doesn't help, its very very likely that the modules doesn't have bootloader installed (we seen failures like that on many HK openlrs modules), and in this case you will need an AVR ISP programmer to flash the bootloader in, USBasp should get the job done and is reasonably cheap.

drak1nguard, 2015-01-14:
I know that i connect it right because of this file http://www.hobbyking.com/hobbyking/store/uploads/648805988X339719X19.pdf 
i did everything right i thing!TO understand the picture better i have an 9xrPRO with http://www.hobbyking.com/hobbyking/store/__27095__OrangeRx_Open_LRS_433MHz_TX_Module_JR_Turnigy_compatible_.html and i try to connect it at http://www.hobbyking.com/hobbyking/store/__27096__OrangeRx_Open_LRS_433MHz_9Ch_Receiver.html with this http://www.hobbyking.com/hobbyking/store/__38395__OrangeRX_USB_Firmware_Update_Kit_for_JR_Futaba_Style_Transmitter_Module.html
Do u see something wrong?If not maybe will be the bootloader!can u link to me a USBasp to buy it??

cTn-dev, 2015-01-14:
That tool you linked -> http://www.hobbyking.com/hobbyking/store/__38395__OrangeRX_USB_Firmware_Update_Kit_for_JR_Futaba_Style_Transmitter_Module.html

isn't compatible with openLRS gear, in this case you should get an cp2102 serial adapter of ebay, they are usually fairly cheap

drak1nguard, 2015-01-14:
can u link one to me that is compatible with my module?

Date: Wed, 14 Jan 2015 03:30:54 -0800

cTn-dev, 2015-01-14:
For example, this one http://www.ebay.com/itm/6Pin-CP2102-Module-USB-2-0-To-TTL-On-STC-To-Arduino-Board-Promini-Download-/281418973953?pt=UK_BOI_Electrical_Components_Supplies_ET&hash=item4185e0c301

drak1nguard, 2015-01-14:
can i use this? http://www.ebay.com/itm/181215336077 for my module?

cTn-dev, 2015-01-14:
that is an ISP programmer, you can use it flash both bootloader and firmware in, but you won't be able to use it with the configurator app (you will have to use either arduino or avrdude)

drak1nguard, 2015-01-14:
can i use the http://www.ebay.com/itm/6Pin-CP2102-Module-USB-2-0-To-TTL-On-STC-To-Arduino-Board-Promini-Download-/281418973953?pt=UK_BOI_Electrical_Components_Supplies_ET&hash=item4185e0c301 u linked to me for the
http://www.hobbyking.com/hobbyking/store/__27096__OrangeRx_Open_LRS_433MHz_9Ch_Receiver.html also to flash it with configurator?

cTn-dev, 2015-01-14:
If the modules have bootloader, yes, if they doesn't you will need both the tool i linked and the ISP programmer

drak1nguard, 2015-01-14:
so if i have http://www.ebay.com/itm/6Pin-CP2102-Module-USB-2-0-To-TTL-On-STC-To-Arduino-Board-Promini-Download-/281418973953?pt=UK_BOI_Electrical_Components_Supplies_ET&hash=item4185e0c301 
AND
http://www.ebay.com/itm/181215336077
i have everything i need to make it work right?

cTn-dev, 2015-01-14:
yup, just make sure to always use them in "3v3" mode (the ICSP have a little jumper on it that you can use to switch it), otherwise you will burn the rfm module on the transmitter/receiver

drak1nguard, 2015-01-14:
i know about the 3.3 volt! do u thing the update kit i used have burned the module always in 3,3 v)....because the servise in HB told me to use it....u are sure it isnt compatible right?

cTn-dev, 2015-01-14:
i am 100% sure that stuff isn't compatible, you are actually a second or 3rd person asking about that kit this month.

About the fact it being burned, i can't really answer that i am afraid :/

drak1nguard, 2015-01-14:
if it is i will ask for a refund!i have the chat with their "specialist" !!!!Can i leave open the ticket so i will ask u for connections when i take the isp?

cTn-dev, 2015-01-14:
Sure, however i would start with the cp2102, if you end up being unable to flash through configurator = there is no bootloader, only after that i would plugin the ISP

drak1nguard, 2015-01-14:
i have bought both and im w8ting to deliver!they are cheap!!! :P ty for your time

cTn-dev, 2015-01-14:
Anytime, good luck with the flashing.

kh4, 2015-01-15:
The OrangeRX flasher is seemingly a PL2303 based USB to serial adapter with some additional circuit, I guess it could be modified to be usable, on the other though pl2303 is like the worst USB2serial chip I have seen ;)

drak1nguard, 2015-01-16:
Didnt understand nothing....!!!dont forget im not a programmer and very new to all this!!!!
cTn-dev they said to me the update kit is compatible.....!!!!!Can u link to me a way other except the chtome to flash the module and a link to the latest update and i will try to use it!!!Plzzzzz....

Transfixus, 2015-08-29:
So I seem to be having the same issue as drak1nguard. I recently bought two additional 100mw Hobbyking openlrs 433Mhz receiver modules. Couldn't flash or connect using FTDI or CP2102. Bought a USBasp programmer and figured out how to burn the bootloader with openLRSng data loaded into Arduino. Then tried again to connect to configurator, no luck...  Uploaded openlrsng data using USBasp up-loader in Arduino successfully and then tried configurator again..... no luck. I started messing with AVRdude to check/change "fuses???" but I got lost real quick. Sigh...  I just want the configurator to connect and smile at me instead of the "10 seconds and no response". (sorry you have to deal with such a newb as me, I really didn't want to bother anyone) Any other suggestions? Did I get in over my head buying these modules?

Nienscecco, 2017-01-06:
Hello guys.  I know this conversation is 2 years-old but I face this exact same situation with the exact same products : 9XR pro, and both TX and RX modules linked above by drak1nguard. 

What was your solution? Were you able to solve the probelm?

Does anyone know what is the needed start msg the configurator needs?  I would be able to program my FTDI chip to send the message to itself using an aux output (CBUS3 for example, I've done that in the past on a work project).

Before flashing a bootloader in my modules, does anyone know how there modules can be used as received? Don't they have a default configuration that is useable right away? Where can I find the info?

Thank you
Thank you

Nienscecco, 2017-01-07:
Hello everyone.  I found my solution.  Simple mistake : I was not connecting DTR because I had seen in the original instructions I read that it wasn't needed.  So the short story is : make sure DTR is connected and make sure RX/TX aren't swaped if you get this error message.

For those interested in a long explanation : 

When you receive your module out of the box, it is empty.  No default firmware.  There must be a bootloader though.  The module needs to be flashed with firmware before you can connect to it because the start message that is sent on the serial port at boot up is part of the firmware, not the bootloader.
 
DTR is connected to the reset pin of the ATMEGA chip inside the module and the configurator needs this signal to reset the module at the right moment so that its bootloader will accept incoming flash programming in a specific protocol. On my custom FTDI adapter I have an LED associated to the DTR pin and I can see that the TX module resets when DTR is pulsed.  

When connecting to the module instead of flashing it, DTR is also needed. Now that the module has a formware, it does send a start message on the serial port when booting up.  The chrome configurator uses DTR to reset the module at the right moment after opening the COM you are using on your computer and listens for the start message at that moment.

csurf, 2018-01-10:
@mikeller 
stale/resolved issue. please close

closed 2018-01-10 by mikeller

snookerdoodle92117, 2022-07-31:
I had something to add to the above thread.  I had the same problem - not being able to connect with the configurator.  What worked for me was initiating a connection with the configurator and power cycling the Tx while it was trying to connect. Everything came up and I was able to setup my receiver also. BTW: I am using OrangeRx OpenLRS 433 w/Bluetooth and the PPM output only receiver.  I pulled the Bluetooth off and used an FTDI.