openLRSng/openLRSng-configurator #20: Read out RSSI using the configurator

frog32, 2015-01-21:
For the purpose of tuning antennas it would be very handy to be able to display RSSI values for both receiver and transmitter in the gui.

csurf, 2018-01-11:
By design, the GUI currently isn't really setup for something like this due to the following:

The way Configurator works is that it goes into 'bind mode' where the TX isn't transmitting; it's simply on standby, and, when in 'spectrum analyzer', it goes into 'receive' mode and samples RSSI on each frequency.

When you go into the RX configuration tab & connect to the RX, it establishes a 2-way connection, for sending&requesting configuration packets. During this connection, the RX is also stuck in bind mode, and it currently doesn't have a means of measuring & sending back the RSSI value (this also applies to the TX). 

In order to print RSSI values for both the TX and RX within Configurator, the following changes would be required:
- add logic on the RX side for a new 'request RSSI' command in 'bind mode' that retrieves the raw RSSI value for the last received packet (the bind mode packet that was just received from the TX)
- add logic on the TX side to also retrieve the RSSI value for the 'response' packet that was sent back from the RX
-  add logic to the Configurator binary protocol (and probably also the CLI protocol) for requesting the two RSSI values, something like 'PSP_REQ_RSSI'. This would instruct the TX to send the 'request RSSI' packet to the RX, and then return the two RSSI values (from steps 1 & 2 above) 
- With that ability, we could potentially setup some kind 'RSSI meter' tab in Configurator that constantly sends RSSI request packets back & forth, and prints out the RSSI values for both the TX and RX in a nice,
 big readout (similar to what you'd see in an OpenTX telemetry page).
-- we could also do stuff like print 'raw, average, max' or use some kind of visual or auditory indicator to show if the RSSI value is improving or worsening over time (maybe like a 'green light'/'red light' color scheme, or play a 'good/bad' beeping sequence ?)

All that being said, there are probably far easier ways to do antenna testing, such as by using the openLRS test script.

@mikeller, what are your thoughts on this?
I have a pretty good idea on how to set this up, but I'm not sure it's a worthwhile effort.

mikeller, 2018-01-11:
@csurf: Doing this through the configurator sounds like a complicated way of doing this. RX RSSI can already be received on the TX through telemetry, and we could look into feeding back TX RSSI into the TX through multiprotocol telemetry as well, so that RSSI can be tested / monitored while in normal operation, instead of only when connected to the configurator.

csurf, 2018-01-11:
I agree, but I'm guessing @frog32 must not have telemetry setup, otherwise he'd have done it that way. @frog32, care to chime in?

I personally don't have a need for this feature (I do antenna testing by other means), so I'm indifferent, but I do have a pretty good of how it could be done in case other users really wanted a means of doing it via the GUI.

csurf, 2018-01-14:
After further review, not worth the effort. closed

closed 2018-01-14 by csurf