openLRSng/openLRSng-configurator #23: Chip not supported, sorry :-(

n8twj, 2015-03-26:
I have been trying to flash OrangeRX RX and the 1 Watt JR TX module... 

I'm using a std FTDI232 dongle, jumpered to 3.3v.  I have the FDTI drivers installed (OS X 10.8), everything seems to be wired correctly, but whenever I attempt to load the firmware I receive:

15:18:55 -- Serial port successfully opened with ID: 30
15:18:56 -- Chip not supported, sorry :-(
15:18:56 -- Serial port successfully closed

Not certain what chip is not supported or what to attempt next.   I literally just received everything from HK and i'm a newb at this stuff...

kh4, 2015-03-28:
Can you check what reads on the top of the biggest IC (CPU). It should read something like ATMEGA328P..

n8twj, 2015-03-29:
It is the correct chip.

Two Rx and one TX unit experience same behavior.   Ordered them all at same
time.

I almost suspected my FTDI might be foobard...but I was able to configure a
Bluetooth board so ?!

Serial monitor at any baud just spews garbage

kh4, 2015-03-29:
It may be that ORangeRX failed to load proper bootloader on them. You need USBASP to load that.

You can use files from http://openlrsng.org/releases/v3.8.2/433/

TX-2-bl.hex for OrangeRX TX unit
RX-3-bl.hex for OrangeRX RX

Notably you should also set fuses to Low: FF High: DE Ext: 05 with USBASP while at that.

n8twj, 2015-04-05:
I have attempted to write the hex using avrdude and hexUploader (which
seems to just be a mac gui for avrdude), but the verification always comes
back as failed.

When I attempt to connect in the Configurator I get:

18:51:05 -- Serial port successfully opened with ID: 8

18:51:08 -- Module - Ope~LRSng TX starti~g 3.8 o~ HW 2

18:51:15 -- Start message not received within 10 seconds, disconnecting

18:51:15 -- Serial port successfully closed

Very similar behavior when I flashed an RX unit with the RX HEX file and
then attempt to connect via Configurator.

TX unit seems to want to work (if i hold button down while powering up, it
goes into bind mode) and I do hear RF interference on my unshielded
speakers when Transmitter is powered on with module plugged in....   But
when I attempt to bind an RX unit, the TX unit never stops beeping nor do
any lights change on RX unit...

Also, i've asked a few local folks and none of us seem to have any clue how
to set the fuses like you say... and my googling has came up with nothing
that looks even remotely familiar to what you provided   :(

csurf, 2018-01-11:
@n8twj 
what's the status on this issue? did you get it resolved? If so, please close this.

@mikeller, close if you feel this is probably a dead issue...

closed 2018-01-11 by mikeller