openLRSng/openLRSng-configurator #25: Compatibility with Gitsly branch of OpenLRSng

doceave, 2015-06-18:
Dearest contributors.

OpenLRSng is by far the most amazing addition to my vehicle. Recently I added Mavlink telemetry using the Gitsly branch of OpenLRSng.

My problem is that Configurator does not seem to be able to connect to the OrangeRx OpenLRS Tx  now that Gitsly firmware is loaded: Gitsly presents it's  version number followed by an error generated by Configurator - "No welcome message received".

As a result I am unable to allow >8 PPM and to map channels to specific ports on my Rx as I did previously.

Could an expert please: 
1) Confirm for me whether or not Configurator is compatible with Gitsly branch of OpenLRSng.
2) If incompatability is the case, direct me to where in the code I can map pins and set PPM  channel number limits.

Thanks so much guys! :)

Amazing work!

cTn-dev, 2015-06-18:
Hi

1. Since gitsly branch is a custom fork of the project i really don't know if its compatible, but from what you are saying it isn't. Configurator is always in sync with the latest officially firmware release.
2. Someone else will have to answer this one, @kh4 ?

doceave, 2015-06-18:
Thanks so much for the reply cTn-dev... I'm certain Kha will have an answer to #2 :)

csurf, 2018-01-11:
@doceave 
We are looking at the possibility of merging the gitsly functionality into the main OpenLRSng code. If that succeeds, it'll be fully compatible with Configurator at that point. Up until then, gitsly remains a separate, custom development fork, so Configurator can't really be expected to support it. 

However, if gitsly ends up having to exist as its own separate firmware, then perhaps we'll have to find a way to integrate support for it into Configurator. 

@mikeller close?

closed 2018-01-11 by mikeller