openLRSng/openLRSng-configurator #27: please feature: full 868 and 915 support  (online fw upgrade)

njozsef, 2015-07-14:
Hi developers.

I working modified 868Mhz orangeRX JR 1W modules and modded 868Mhz receivers.
My problem: configurator support only 433Mhz (fw upgrade). Please compile and upload all versions and/or
please allow a choice for example dop-down box (433/868/915).
This feature very facilitate the upgrade.

if this solution is hard work, please upload all firmware the firmware folder

Thank you for professional work.
Best regards. Jozsef.

kh4, 2015-07-17:
All firmwares are available here

http://openlrsng.org/releases/

csurf, 2018-01-11:
@mikeller, does this sound reasonable? We'd have to package ALL of the firmwares for each type of module (433, 868, 915) and for each type of hardware, unless we change over to some kind of system like you guys do in betaflight where the FW is downloaded on demand... thoughts?

mikeller, 2018-01-11:
@csurf: That is exactly the idea. There is already a change to configurator (#36) that uses the GitHub API to check for new configurator releases. We'll be leveraging the same API to check for the latest firmware release, and then download that from GitHub.

csurf, 2018-01-11:
do you guys have a nice 'template' for the 'download & verify' routine that we can more/less 'plug&play' into Configurator

mikeller, 2018-01-11:
I plugged whatever was directly pluggable in #36. I want to do the selecting / downloading part different to how Betaflight does it, letting the user only select the version, then downloading all of the hex files, and caching them locally, so I'll have to rewrite most of that part.