openLRSng/openLRSng-configurator #32: Allow use of directly connected receiver for spectrum analyzer function.

gweep, 2016-11-14:
For some reason for which I don't understand (but can live with) all access to a receiver for programming purposes needs to be through a bound transmitter.  Again, I don't understand it but it is not a problem.

However, I see no reason not to permit a serially connected receiver to perform the spectrum analyzer function.   This would be convenient as the RX serial port connections are easier to get to that those on the TX module.   

Currently, which device provides the spectrum data, the serially connected TX or the bound RX?

thanks

csurf, 2018-01-11:
When connected to the TX, the spectrum analyzer comes from the TX (it goes into 'receiver mode' and just listens & samples RSSI across the selected frequency band).

If you want to run the spectrum analyzer from the RX, you have to put a jumper across a couple of the output pins on the RX (depends on the RX hardware), and connect from Configurator to the RX (over serial, using an FTDI adapter or similar). Configurator should automatically detect that it's connected to an RX that's been forced into 'scanner mode' and immediately opens up the 'spectrum analyzer' tab. At that point, what you see in the spectrum analyzer is from the perspective of the RX.

I had worked on the ability to direct-connect to the RX a while back. It requires adding a bunch of code to the RX side for handling the binary protocol that's used to communicate with Configurator (this protocol is currently only implemented within the TX code). I actually got it to work but kinda gave up because I didn't think it would be accepted into the main code branch seeing as it was probably somewhat of a novelty/niche feature.

So, It's definitely possible, but does add bloat and complexity to the RX side.
@mikeller, thoughts?

mikeller, 2018-01-11:
@csurf: I can see where @gweep is coming from. But, as he says, the functionality is already there (by connecting through the TX), so this is just convenience. I'd leave this feature request around, if somebody wants to work on it, but I think there are things with higher priority to spend time on.

csurf, 2018-01-12:
@gweep 
ok, I dug up my old code, and I'm working on a couple of patches for both the firmware & GUI to get this working. 

Only problem so far is getting the frequency ranges for the 'beacon frequency' selection. The RFM frequency type is stored in the TX configuration. The RX code doesn't currently have a way of 'knowing' what frequency range it operates on (433, 868, or 915). 

Configurator relies on getting the frequency info from the TX configuration data. However, if you're directly connecting to the RX, this info isn't available, so it doesn't know what frequencies to allow for the beacon frequency. Consequently, any number you put in gets highlighted as being invalid.

One possible solution would be to try to integrate the RFM type into the first two bits of the 'type' field of the RX configuration.

csurf, 2018-01-22:
@gweep 
this has been added in #45 and openLRSng/openLRSng#165.
testing required

closed 2018-01-23 by csurf