openLRSng/openLRSng-configurator #42: 'MULTI' setting in 'serial port speed' does not persist in backup

mikeller, 2018-01-11:
When setting the serial port speed for the TX module and selecting 'MULTI' (for Multiprotocol), the setting persists normally to the module.
But when backing up the profile into a file, and then restoring from the file, the serial port speed is set to '115200'.

csurf, 2018-01-12:
@mikeller 
...just an FYI, PR #43 which might be somewhat related

csurf, 2018-01-12:
@mikeller 
ok, this looks like it's due to the tx_console_baud patch.
Are you testing off firmware master?

csurf, 2018-01-12:
if we upgrade the firmware to 4.0.0 and use the patched Configurator from #40, it *should* work

csurf, 2018-01-12:
ok, I tested against firmware master with version upgraded to '4.0.0' and with Configurator w/ #40, and was able to successfully backup & restore a profile.

FYI, it wasn't just the MULTI profile by this bug. 
The entire TX config can't be saved due to being incompatible. 

I need to set the TX config into 'read-only' mode for older config versions...

csurf, 2018-01-23:
this should be resolved by #46, and latest 3.9 firmware from repo should work / be compatible with Configurator. closing. please reopen if the problem persists...

closed 2018-01-23 by csurf

mikeller, 2018-01-23:
Confirmed working, thanks @csurf.