openLRSng/openLRSng-configurator #53: Can't connect to transmitter --

linuxslate, 2018-02-03:
Using the Configurator from the Chrome Web Store, I can connect to my transmitter fine:

```
17:03:11 -- Running - OS: Linux, Chrome: 64.0.3282.140, Configurator: 0.57
...
17:10:23 -- Serial port successfully opened with ID: 7
17:10:28 -- Module - OpenLRSng TX starting
17:10:28 -- Requesting to enter bind mode
17:10:28 -- Transmitter Firmware version - 3.8.8
17:10:28 -- Transmitter BIND data received
```

TX Tab is enabled -- works fine.

-------------
Using Configurator: 1.0.0-rc1, I cannot connect

```
17:13:00 -- Running - OS: Linux, Chrome: 64.0.3282.140, Configurator: 1.0.0-rc1
17:13:00 -- Please click on "Request Optional Permissions" button to grant application required USB access
17:13:02 -- Loaded release information for configurator from GitHub.
17:13:06 -- Optional USB permissions granted
17:13:10 -- Serial port successfully opened with ID: 8
17:13:14 -- Module - OpenLRSng TX starting
17:13:14 -- Requesting to enter bind mode
17:13:17 -- Communication through Phoenix Serial Protocol was never established, connecting failed
17:13:17 -- Serial port successfully closed

```

At this point, the TX seems to be in bind (flashing green light), and it will bind with the RX, but  it has still timed out, so there is nothing I can do.

Also note that Configurator: 0.57 won't work with firmware newer than 3.8.8, so I am basically stuck with 3.8.8 for now.

csurf, 2018-02-05:
@linuxslate 
it's a timing issue.
I can provide a patch for you to test, or you can try hacking into the javascript code if you feel comfortable.

csurf, 2018-02-06:
fixed by #54

closed 2018-02-06 by csurf