openLRSng/openLRSng-configurator #55: Unable to access tabs using Master:ae933b1 and 3.9.0-rc2

NopeDK, 2018-02-11:
15:21:11 -- Running - OS: Linux, Chrome: 64.0.3282.140, Configurator: 1.0.0-rc1
15:21:13 -- Serial port successfully opened with ID: 1
15:21:13 -- Using cached release information for configurator releases.
15:21:14 -- Module - OpenLRSng TX starting
15:21:14 -- Requesting to enter bind mode
15:21:14 -- Transmitter Firmware version - 3.9.0
15:21:23 -- You can't view this tab at the moment. You need to connect first

[image]

csurf, 2018-02-11:
@NopeDK 
Try waiting until it enters bind mode before clicking on the tabs.
It will time out if it can't enter bind mode, and then you can try reconnecting again.
Sometimes you might need to try a couple-few times to get it to work.

If it consistently times out, then it's a timing issue.
I had already fixed it, but the port timing can vary on different computers & with different TX modules.

Try that and report back.

NopeDK, 2018-02-12:
@csurf
Retrying now, currently at 30 minutes without any change.
Seems to be stuck on "Transmitter Firmware version".
If I disconnect and try again I get the exact same thing.

NopeDK, 2018-02-12:
Increased timeout on line 327 from 250 to 2500, no change in output.

NopeDK, 2018-02-12:
Increased timeouts on line 325-329. Now the error has changed to "Communication through Phoenix Serial Protocol was never established, connecting failed" followed by immediate serial port closure.

csurf, 2018-02-12:
what type of transmitter hardware are you using?

NopeDK, 2018-02-12:
OrangeRX 9ch RX for both TX and RX.

csurf, 2018-02-12:
ok, I've partially identified the issue (firmware compatibility problem) and I'm working on it.

Would you be able to build the firmware with Arduino IDE? firmware master should work.

NopeDK, 2018-02-12:
I am setting the environment up now.

NopeDK, 2018-02-12:
Downloaded Master:948fc57 of the firmware and flashed to both TX and RX. It now works with the titled version of the configurator.

csurf, 2018-02-12:
ok, good. pls let me know if you continue to have connection problems.

I'll leave this issue open because Configurator still needs to be fixed.

mikeller, 2018-02-19:
@csurf: Why was this closed?

Daniel-1276, 2018-08-30:
Hi, it seems current FW release "openLRSng 3.9.0-rc2" still has this issue. @NopeDK could you may share a flashable binary for TX type 2 and RX type 3 please ? No idea how to setup Arduino build environment to do it myself.

Daniel-1276, 2018-08-30:
Managed to compile by myself from the current master repository ... Issue solved :) Heading to multi-protocol feature now :)

mikeller, 2018-09-01:
@Daniel-1276: You got a point there - no reason to withhold the improvements from users due to lack of time to write nice release notes, I did a quick-fire release from latest in master: https://github.com/openLRSng/openLRSng/releases/tag/3.9.0-rc3