openLRSng/openLRSng-configurator #60: master version settings cannot be saved "compatibility mode"

JamesCordell, 2018-05-12:
Hi

The Configurator produces the message

"Configurator is running in compatibility mode, you will have to update your firmware to the latest version before trying to save any of of the settings"

And does not permit settings to be saved for the TX. This appears to be a versioning issues. I see the firmware is compiled into the configurator, so i'm not sure where the issues is. 

This is not an issue with the development version of the configurator 5.4 with the fw version 3.9.0 

I'm not sure how to build the fw or I could debug this. I am using DTF UHF Delux

mikeller, 2018-05-13:
This is to be expected. In previous versions of the firmware and configurator, the firmware was distributed with the configurator. We'll have to change this, and change how the versions are matched.

For now, use the version of the configurator from the repository when using the firmware from the repository.

antaljani, 2021-10-19:
Is there any solution for this topic? I run the same problem. Configurator is a newer version than the firmware. How you can update the firmware then??? This is not so logical...